DanielCoelho112/synfeal #56: Collect datasets using KinectFusion

DanielCoelho112, 2022-05-19:


DanielCoelho112, 2022-05-19:
Hi @miguelriemoliveira and @pmdjdias,

I use this [repo](https://github.com/JingwenWang95/KinectFusion) to run KinectFusion on our data and the results are quite bad. Look at the reconstructed model:

[image]

The algorithm isn't able to understand that the walls are orthogonal, and thus produce an almost flat structure. The poses are the following:

[image]

To understand these poor results I did some research and found a paper called: Metrological evaluation of KinectFusion and its comparison with Microsoft Kinect sensor.  They state some problems of kinect fusion, such as:

_The usage of depth image based registration technique (ICP) causes large errors
when camera motion is large or scene is poor in 3D structure (i.e. flat regions). And
voxel based scene representation is problematic for reconstruction of a large area due to
memory limitations_ 

Their limitations are exactly our case. When we are looking at the boiler, the algorithm is not that bad, the problem is when the camera moves towards the flat wall...  Furthermore, our lab is really big compared with the rooms where kinecfusion was used in the 7 scenes:

[image]

In addition, the algorithm KinectFusion was developed by Microsoft, and thus there are few public implementations. The majority of implementations only work on windows and are real-time, meaning we have to use a real Kinect. 

I'm running out of ideas here, it looks that current data collection techniques are accurate enough but only work in small environments with a lot of objects.

miguelriemoliveira, 2022-05-19:
Hi @DanielCoelho112 ,

do not despair. We are on the right track, although sometimes we must cross a desert pass (I have been reading mystical literature recently and then you suffer : ) )

How about shortening the distance steps you take from frame to frame? Perhaps that's what's going wrong with fusion, the images are very far away. Notice it works on kinect, which has 30Hz.

Another option: Can you try out rtab map? That could work and it has a lot of parameters ...

DanielCoelho112, 2022-05-19:
Hi @miguelriemoliveira,

> How about shortening the distance steps you take from frame to frame? Perhaps that's what's going wrong with fusion, the images are very far away. Notice it works on kinect, which has 30Hz.

Already did that, I increased the number of frames between the source and target poses, but I'll try with a bigger number again.

> Another option: Can you try out rtab map? That could work and it has a lot of parameters ...

I thought about it, and we could give it a try. The problem is that, as far as I know, for acquiring the ground truth in localization no one uses slam... But if we cant use anything else, maybe is our best option.

DanielCoelho112, 2022-05-19:
Yes, you're right. So it makes even more sense. Tomorrow I'll do it.

DanielCoelho112, 2022-06-16:
Kinect Fusion is unable to work on our datasets due to the reasons mentioned above.

closed 2022-06-16 by DanielCoelho112