DanielCoelho112/synfeal #57: Create datasets with variable lighting conditions

DanielCoelho112, 2022-05-19:
To increase the robustness of our models with respect to various lighting conditions throughout the day, we should produce datasets with variable lighting conditions.

DanielCoelho112, 2022-05-19:
Hi @miguelriemoliveira,

to give continuity to my desert pass, this is proving to be harder than I thought.

So we have two options to change the illumination:

1. through services
2. through topic

**Services**

I create a [script](https://github.com/DanielCoelho112/localization_end_to_end/blob/main/localbot_core/scripts/set_lights#L9) that changes the illumination via SetLightProperties service. The service works in a way that if I do SetLIghtProperties, and then GetLightProperties, the information retrieved is the one that I sent. The problem is that in gazebo nothing changes...

**Topic**

There is a topic called "/gazebo/my_room_024/light/modify" that we can use to change the lighting conditions. When I do:

    gz topic -i /gazebo/my_room_024/light/modify

I get this:

    Type: gazebo.msgs.Light

The problem is that I cannot import that message... The documentation of that msg is the following:
https://osrf-distributions.s3.amazonaws.com/gazebo/msg-api/dev/light_8proto.html

I read somewhere that I need to build that message, but I have no idea how to do it, do you have any ideas?
I looked around and I found this:

https://github.com/arpg/Gazebo/blob/master/gazebo/msgs/light.proto

I was thinking about creating our own custom message, but the extension of this file is .proto, and I don't know what to do with it.

pmdjdias, 2022-05-23:
Nice, looking forward to see the impact on training ;-)!

miguelriemoliveira, 2022-05-23:
Yes, I also like it. One light I think is missing is the outside directional light of the sun. 

In any case I think this looks very nice: we now have these highly illuminated walls which are actually very realistic, Lucas had to take some real images that looked very much like this from the color correction.

DanielCoelho112, 2022-06-08:
This is done.

closed 2022-06-08 by DanielCoelho112