DanielCoelho112/synfeal #58: Collect datasets using RTAB

DanielCoelho112, 2022-05-26:


DanielCoelho112, 2022-05-27:
Hi @miguelriemoliveira and @pmdjdias,

I already applied rtab and I think the results look promising:

[image]

We can see a slight difference between the gazebo ground truth and rtab. If we check the 4x4 pose of the last frame, we get this:

Gazebo:

        0.35333,-0.17789,0.91843,2.09022
        -0.93422,-0.01579,0.35635,3.38141
        -0.04888,-0.98392,-0.17177,0.04658
        0.00000,0.00000,0.00000,1.00000

rtab:
        
        0.34663,-0.18187,0.92020,2.07315
        -0.93659,-0.01329,0.35017,3.31657
        -0.05144,-0.98323,-0.17495,0.10495
        0.00000,0.00000,0.00000,1.00000

The position error oscillates between 2 to 7 cm.

I believe that with these results, we will already notice some differences in the learning of the models.  However, since rtab has a million hyperparameters to tune, I was thinking I could explore them a little. What do you say?

miguelriemoliveira, 2022-05-27:
Great news! Good idea to explore rtab a bit more.

But perhaps you can have several flavors of rtabmap, to see if we get some more columns in the dataset.

How about revisiting the v2o using images which are closer together?

DanielCoelho112, 2022-05-27:
> Good idea to explore rtab a bit more.

I did some experimentations, and now we have 3 versions of RTAB.
I produced a dataset with 160 frames with variable ilumination and then ran the 3 versions.

**1: default**

[image]

This image compares the position and rotation error of the rtab poses compared with the ones provided by gazebo.

This is the best rtab. The error is stable and around 2.5 cm.

**2: change matches  2D-3D to 3D-3D**

[image]

In this case, the error is accumulating throughout the frames. The magnitude of the error is quite big.

**3: change graph optimizationg algorithm to TORO**

[image]

here the error is also growing but at a lower speed. This image is good to illustrate why current techniques aren't accurate enough - accumulative error.

I think we could use the 3 versions for our table.

> How about revisiting the v2o using images which are closer together?

I will give it try. We could also try visual slam, right? Because rtab uses depth...

miguelriemoliveira, 2022-05-28:
Hi @DanielCoelho112 ,

these results are very good. Congratulations. 

> I will give it try. We could also try visual slam, right? Because rtab uses depth...

I think v2o is visual slam.

BTW, are you sure the knect fusion would not work a bit better if you give it a images with less spacing?

DanielCoelho112, 2022-06-02:
Hi @miguelriemoliveira and @pmdjdias,

Running RTAB with 8000 frames is not possible due to 2 reasons:

1. High memory consumption;
2. The cumulative error leads to abnormal results.

To surpass this, I ran RTAB with batches. The procedure was to run RTAB continuously until the results start looking bad, then I ran RTAB again starting from that frame. 

By doing this, we are giving a lot of help to RTAB because we are using privileged information to assess the performance of RTAB.

The error across frames is the depicted here:

[image]

The mean error is 0.11 cm. The largest errors in RTAB are with frames facing the walls without objects, which is a problem reported in the literature.

Tomorrow I'll validate our main thesis.

miguelriemoliveira, 2022-06-03:
Hi @DanielCoelho112 ,

>  The procedure was to run RTAB continuously until the results start looking bad, then I ran RTAB again starting from that frame.

what is  the exact criterion?

It looks like you're heading in the right direction. Keep it up.

DanielCoelho112, 2022-06-03:
I didn't use an exact criterion.
Imagine if the errors per frame are:  0.07,0.04,0.10,0.02,0.5,1.2,2.0. In this case, I would start RTAB again from the frame with an error of 0.02.  I always tried to restart from a frame with low error to help RTAB. 

This procedure leads to batches with variable sizes, for example, the first batch had 1800 frames, whereas the second batch had 200 frames.

miguelriemoliveira, 2022-06-03:
OK, but for the paper we should have a rigorous criteria , either restart when you fall bellow a maximum error, or a fixed batch size, or a combination of several.