DanielCoelho112/synfeal #60: Additional outputs missing in PoseNet

DanielCoelho112, 2022-06-03:
Hi @miguelriemoliveira and @pmdjdias,

When training PoseNet I was experiencing a lot of overfitting that not even high dropout rates and high L2 regularizations were able to reduce.  Today I went to check the PoseNet paper again and found that they were still using 2 additional outputs to stabilize the google net.

The github implementations of PoseNet that I found didn't consider this. I think we can achieve even better results with this feature. I'll implement it and let you know if it helped.

miguelriemoliveira, 2022-06-05:
Hi @DanielCoelho112 ,

I hope that is the cause for te over fitting. Let me know how it goes.

DanielCoelho112, 2022-06-08:
Done. It helped solve the overfitting.

closed 2022-06-08 by DanielCoelho112

miguelriemoliveira, 2022-06-08:
An image of the training please !!!!

DanielCoelho112, 2022-06-08:
[image]

Although it looks like there is still some overfitting the reason is the following:

Loss_train = Loss_final + 0.3(Loss_auxiliar1 + Loss_auxiliar2)

Loss_test = Loss_final

So, in training, we are adding 3 negative terms, and in test, we are only adding one.  I say that we are free of overfitting because our test loss keeps decreasing and following test loss.

miguelriemoliveira, 2022-06-08:
Looks good. Actually, it looked like you could've continued the training a bit more.

pmdjdias, 2022-06-08:
Agree… “Uma imagem vale mil palavras” 😉!

--Paulo