DanielCoelho112/synfeal #65: Implement PoseLSTM

DanielCoelho112, 2022-06-17:
paper: https://openaccess.thecvf.com/content_ICCV_2017/papers/Walch_Image-Based_Localization_Using_ICCV_2017_paper.pdf

DanielCoelho112, 2022-06-17:
Hi @miguelriemoliveira  and @pmdjdias,

based on this image:
[image]

Do you think the variable **z** is referring to the size of each rectangle, or to the size of the sum of the 4 rectangles?

miguelriemoliveira, 2022-06-17:
Hi @DanielCoelho112 ,

really not sure. 

I would say it is not a size. That would be a scalar.  It should be the weights of the network arranged in several dimensions.
For example y is the typical output of a CNN, where, at  the image resolution is collapsed and represented "semantically" by a vector of very high dimensionality, i.e. a vector of 2048 weights in this case.

I would say the z is the same. From y to Y there appears to be the inverse process of decoding the vector y, gaining back some semantic resolution (my term) of 32x64. 

Then from Y to z again there is an encoding by collapsing again the "semantic resolution" back to a semantic vector of dimension 128.  So z is not the size, it represents the 128 values of the weights which are the semantic representation of the input data at that layer of the network.

My experience with this is not much so I maybe completely off here ... it is the most likely scenario : - )

DanielCoelho112, 2022-06-18:
Hi @miguelriemoliveira,

yes, you're completely right.  The main difference between this approach and PoseNet is how to handle the 2048 feature vector. PoseNet applies FC directly whereas PoseLSTM employs a structured dimensionality reduction using recurrent units (LSTMs). 

> Do you think the variable **z** is referring to the size of each rectangle, or to the size of the sum of the 4 rectangles?

Sorry, I didn't express myself correctly. I agree with you that z represents the 128 weights, my question is what is z in the figure. Is it each rectangle, or it is the concatenation of the rectangles? 

[image]
[image]

To figure this out, I implemented both alternatives and the one that achieved the best results was the first option: z represents each rectangle, meaning that the final feature vector is of shape (1, 512).

miguelriemoliveira, 2022-06-18:
Great. What I still do not understand very well is how these authors come up with these ideas. It continues to sound all very arbitrary ...

DanielCoelho112, 2022-06-18:
Actually, these authores explain why they used this architecture. Their goal is to fight the overfitting problem of posenet. If you want we can discuss their idea later. Although the idea is simple, it involves complex concepts, as is the case of LSTMs.

But I agree with you, it is very arbitrary. For instance, it makes sense to tranform the 2048 vector into a 2 dimensional matrix, but why 32x64? In the paper they didn't even mention that shape...

miguelriemoliveira, 2022-06-20:
OK, lets discuss later, but only by the end of the week. This is a busy one.

DanielCoelho112, 2022-06-20:
Results are updated in #61.

closed 2022-06-20 by DanielCoelho112