DanielCoelho112/synfeal #69: Download 3D model of a indoor scene

DanielCoelho112, 2022-06-20:
To prove that Synfeal works in other datasets, we should download a 3D model.

DanielCoelho112, 2022-06-20:
Hi @miguelriemoliveira and @pmdjdias,

I found these 3D models:
[image]
[image]

I think the first one is quite good for us, what do you say?

pmdjdias, 2022-06-20:
It would be better if we could use uor models... I have some from my time in Italy... The geometry of the firste does not seem very good (seems like the ceiling is kind of "wavy"...
[image]

DanielCoelho112, 2022-06-23:
We will advance with the first model due to its geometry.

closed 2022-06-23 by DanielCoelho112