DanielCoelho112/synfeal #72: Implement a data collection technique suitable for 3D models with more than 1 room

DanielCoelho112, 2022-06-24:
Hi @miguelriemoliveira,

I thought about this and, based on the time we have to submit the paper, I think I cannot come up with a solution that works for any dataset.  However, we can use the existing techniques if we divide the 3D model into two models:

[image]

We can collect two separate datasets, and then merge them. The downside of this approach is that we won't have a path from one room to the other, but I don't think that is a big deal...

DanielCoelho112, 2022-06-27:
Hi @miguelriemoliveira,

we can use this issue for your implementation of the data collection with restrictions.
Let me know when you're available to discuss the technical details.

miguelriemoliveira, 2022-06-27:
It looks great. Congratulations!

One suggestion: the video would look so much better if you could show the points of start, and the randomized points B, with red color if they caused collisions and green if they did not.

DanielCoelho112, 2022-06-27:
> the video would look so much better if you could show the points of start, and the randomized points B, with red color if they caused collisions and green if they did not.

Yes, that would be nice. We would have to use RViz instead of Gazebo for that visualization, but it isn't anything new... I think we should accelerate the fundamental things, and then when the models are training, I'll do it.

> just another idea. You could also do another check for each P2, which is to check collisions with a ray cast from P2 along the camera's optical axis. If there is a collision too close to the camera that would mean that we have an image which is probably not good like those we had in 0.24 near the wall.

This condition solves some of those cases : 
>> I also added another condition to guarantee that P2 is at least 20 cm apart from the collision.

[image]

However, there are cases where it fails... Furthermore, the problem, as you said, isnt being too close to the wall, is being close to the wall along the camera axis. I will remove the condition I did, and I'll implement your idea.

DanielCoelho112, 2022-06-28:
Hi @miguelriemoliveira,

 the raycast applied to P2 is already implemented. I used a distance of 1.2 meters (another parameter to the model3d_config yaml file)

miguelriemoliveira, 2022-06-28:
Great

DanielCoelho112, 2022-06-30:
Hi @miguelriemoliveira,

check this:

https://uapt33090-my.sharepoint.com/:v:/g/personal/danielsilveiracoelho_ua_pt/Ee0q4n43dDVKj02Tb3h-OUYBBXRxj_vQGkO4JLXZer-xdA?e=vGMOe5

Whenever a collision occurs, I added a sleep of 2 seconds, otherwise, we wouldn't have time to see the red marker.
I also added a yellow marker when the p2 is too close to a wall.

I think it looks nice, what do you say?

miguelriemoliveira, 2022-06-30:
Much better. Very nice!

closed 2022-07-01 by DanielCoelho112