DanielCoelho112/synfeal #77: Create a dataset using Santuario as the 3D model

DanielCoelho112, 2022-06-27:


miguelriemoliveira, 2022-06-29:
> Hi @miguelriemoliveira <https://github.com/miguelriemoliveira>,
>
> using 18000 frames we get this:
> [image: image]
> <https://user-images.githubusercontent.com/73063947/176495620-c0121af4-0adc-47b6-8c7c-7a71c66cb3a5.png>
>
> I think we should leave this 18000 for training, and collect now 4000 for
> testing.

Very nice. Agree, extra 4000 for test. But those should cover the entire
space...

I think I can improve rhe visualization a bit. Can you let me know how you
visualize so i can try a few tricks?

DanielCoelho112, 2022-06-30:
Hi @miguelriemoliveira,

first download seq22 from[ here](https://uapt33090-my.sharepoint.com/:f:/g/personal/danielsilveiracoelho_ua_pt/Eq-KV5MbmtpOo9PdgcWkZVkBlmkLszmjil0ZWXptYspIyA?e=4va6LF) 

then run:

    roslaunch localbot_localization visualize_dataset.launch dataset:=seq22

seq22 is from the lab 024, so you may have to change the directory in the visualize_dataset script

PR #84 (2022-07-08): Create a dataset using Santuario as the 3D model
Fixes #77
commit d892e3f9 (2022-07-08): Create a dataset using Santuario as the 3D model
Fixes #77
closed 2022-07-08 by miguelriemoliveira