DanielCoelho112/synfeal #86: Online database of 3D models

andrefdre, 2022-11-08:
Need to store the 3D models of the rooms in a server so it can be easly shared, since they are large files.

DanielCoelho112, 2022-11-08:
Hi @andrefdre,

yes, I'm thinking about using DeepLar to store the 3D models. We can use Rsync to synchronize the files, but I've never done this before. Let's see what @miguelriemoliveira says.

For now, download the 3D models from this link: https://uapt33090-my.sharepoint.com/:f:/g/personal/danielsilveiracoelho_ua_pt/EpBbrcKtRh1Lm17gAk6767sBSNLIkUoGvIBtdhSvEGB8_Q?e=crkCWw

Beware that the 3D models have to be in the following directory: /home/user/models_3d/localbot/...

If you have any questions let me know.

miguelriemoliveira, 2022-11-08:
Hi,

We implemented a script to do this in the atom datasets.

https://github.com/lardemua/lars_datasync

Perhaps it helps?

pmdjdias, 2022-11-09:
We can also use the space we have in arca
\\arca.ua.pt\Grupos\IEETA\VARLab
But I need to ask acesse to your UA user (do not remember how to do it). If you think it can be useful, can check.

DanielCoelho112, 2022-11-10:
Hi,

to push new changes to the models_3d folder run the following:

    rsync -avrz --delete /home/<user>/models_3d/ <ssh_user>@193.137.172.109:/home/danc/models_3d

to pull:

    rsync -avrz --delete <ssh_user>@193.137.172.109:/home/danc/models_3d/ /home/<user>/models_3d/ 

To simplify you can put this into sh files and then run aliases.

@miguelriemoliveira can I create a user in DeepLar for @andrefdre? Or is it better for me to ask permission from prof. Vitor?

miguelriemoliveira, 2022-11-10:
Ask for his permission. It is better.

closed 2023-12-12 by DanielCoelho112