DanielCoelho112/synfeal #88: Collecting data with random mode gives error File type: dae not supported

andrefdre, 2022-11-10:
When tying to capture data with the mode automatic gives the error File type: dae not supported. But with interactive works.
Also it gave the error saying trimesh was needed, which I installed but wasn't in the requirements.txt.

DanielCoelho112, 2022-11-10:
Hi @andrefdre,

you're lucky I'm answering before @miguelriemoliveira :)

When posting errors, you should first post the command you ran, and then the error itself.
Check this example https://github.com/lardemua/atom/issues/535.

About trimesh, don't forget to update the requirements.txt.

andrefdre, 2022-11-10:
Upsi, the comannd I ran was 
```
rosrun synfeal_collection data_collector -nf 100000 -m random -mc 'santuario.yaml' -s name_of_the_dataset
```
and got the error:
```
Traceback (most recent call last):
  File "/home/andre/catkin_ws/src/synfeal/synfeal_collection/scripts/data_collector", line 122, in <module>
    main()
  File "/home/andre/catkin_ws/src/synfeal/synfeal_collection/scripts/data_collector", line 72, in main
    data_collector = AutomaticDataCollection(
  File "/home/andre/catkin_ws/src/synfeal/synfeal_collection/src/automatic_data_collection.py", line 72, in __init__
    self.mesh_collision = trimesh.load(
  File "/home/andre/.local/lib/python3.8/site-packages/trimesh/exchange/load.py", line 151, in load
    raise ValueError('File type: %s not supported' %
ValueError: File type: dae not supported
```

About the trimesh I will add it to the requirements.txt then.

miguelriemoliveira, 2022-11-10:
> you're lucky I'm answering before @miguelriemoliveira :)

Right :-)

DanielCoelho112, 2022-11-10:
I think I got it: https://github.com/mikedh/trimesh/issues/439

This should solve it: 

    pip install --upgrade pycollada

If it does, another dependency...

andrefdre, 2022-11-10:
Thanks it solved my problem with:
```
pip install --upgrade pycollada
```
I will add it to the dependencies.

closed 2022-11-10 by andrefdre