DanielCoelho112/synfeal #89: New mesh of Mercado Negro

andrefdre, 2022-12-21:
The scan of the small room made in Mercado Negro was added to Synfeal. Since it isn't watertight, a collision mesh smaller than the room was implemented to prevent the camera leaving the scenario.
The mesh: 
[image]

Results obtained after training the model:
[image]
[image]

Steps to achieve the results mentioned before:
- Create the mesh from a point cloud.
- Create the .world file and restrict the camera's movement to a box smaller than the scenario.
- Capture RGB images, around 500 for the training dataset and around 250 for the test dataset.
- Process the images with the -f flag, since the images are RGB.
- Train the model with the posenetgooglenet, 30 epochs and a batch size of 10.
- To produce the results, the model obtained was tested again with the test dataset.

While commenting the results with @DanielCoelho112 he mentioned that the loss of the test dataset wasn't lowering as expected, mentioning that a low batch size can be one of the culprits. Unfortunately, my graphics card doesn't have enough memory to train with a bigger batch size.

DanielCoelho112, 2022-12-21:
Hi @andrefdre,

nice description. Can you also post the commands you used to collect the different datasets,  and the command to run the training?

Share the datasets with us, you can use the deep lar, or onedrive. 

About the batch size, not sure if I said that, maybe it was the beer talking 😅. I think the problem is somewhere else.

miguelriemoliveira, 2022-12-23:
Hi @andrefdre ,

nice work. I thik 500 images for training are a very small amount, which may lead to the difference between training and test.

andrefdre, 2023-02-08:
I trained a new model in DeepLAR with a training dataset of 50k images and testing dataset with 12.5k I used Hourglass model which is different from the first train mentioned before. The model trained for 70 epochs, but after 20 didn't improve.
[image]
The loss is better than the previous model with those two occasional spikes. 
[image]
@miguelriemoliveira I also already reed the thesis the teacher suggested and have some questions on how it will influence this work.
When is it possible to have a meeting to discuss next steps?

andrefdre, 2023-02-14:
After speaking with Lucas he remembered me that he spoke with Mercado negro through Instagram, I sent a message Friday but still haven't got an answer.