DanielCoelho112/synfeal #93: Next steps

andrefdre, 2023-02-28:
I spoke with @DanielCoelho112, and he is busy until next week. What should I focus meanwhile @miguelriemoliveira ? 

He gave me some stuff to look around:
- Why the current model takes so much memory
- Check if a bigger dataset will perform better 

Should I move on to adding objects?

miguelriemoliveira, 2023-02-28:
Hi @andrefdre ,

I still think there is something wrong with the training and that we need @DanielCoelho112 to look into it or do an example with you.

So my suggestion is that you move to the objects for now, and then come back to this.

Send your phone number to my email and I will call.

pmdjdias, 2023-02-28:
I agree, start to adapte the Lighting code to movement until Daniel can help to overvcome the problems in the training seems the way to go.

miguelriemoliveira, 2023-03-01:
https://doc.babylonjs.com/toolsAndResources/utilities/IsInside

andrefdre, 2023-03-01:
Based on the suggestions made by professor @miguelriemoliveira I manage to find this in Open3D which I think is what I want http://www.open3d.org/docs/latest/tutorial/geometry/distance_queries.html. I will close this issue, since now I will be focusing in implementing objects in the scene.

miguelriemoliveira, 2023-03-01:
Read through it and it looks good. Try to implement it and make a test to see if you can randomly generate 20K points and then paint green the ones that are inside and red the ones that are outside.