DanielCoelho112/synfeal #94: Implement Object movement in the scene

andrefdre, 2023-03-01:
I will now focus on implementing objects moving in the scene when the camera moves:
- [ ]  Find if object are inside the mesh (Found this http://www.open3d.org/docs/latest/tutorial/geometry/distance_queries.html)
- [ ] Generate random poses for objects
- [ ] Check if objects are touching the ground
- [ ] Move objects in a path

andrefdre, 2023-03-02:
I manage to generate random points and perform a check to see if the point is inside the mesh. I used `result = self.mesh_collision.contains(point)` from trimesh and I based on this post mikedh/trimesh#241.

Short video of a random cube being moved in the scene:

https://user-images.githubusercontent.com/58526188/222509410-dfb95bbd-4988-4e5c-9657-1e3729dc945b.mp4

Right now it's only using the point of the pose, so part of the cube is inside the walls. I could also send an array of points to .contains() so in this case I could send the 4 corners.

miguelriemoliveira, 2023-03-02:
Hi @andrefdre ,

> I manage to generate random points and perform a check to see if the point is inside the mesh. I used result = self.mesh_collision.contains(point) from trimesh and I based on this post https://github.com/mikedh/trimesh/issues/241.

So you are using trimesh, not open3d? 

congrats. The video  looks good. 

> Right now it's only using the point of the pose, so part of the cube is inside the walls. I could also send an array of points to .contains() so in this case I could send the 4 corners.

I am not sure how long it takes, but the best would be this: The object you want to add to the scene is also a mesh, so test all the triangles to see if all are inside the room.

If this takes too long because the object mesh has too many vertices, you can create a convex hull mesh of the object and then test only the fewer vertices of the convex hull.

andrefdre, 2023-03-06:
> So you are using trimesh, not open3d?

Yes I decided to use trimesh since we already use it to check collision.

> I am not sure how long it takes, but the best would be this: The object you want to add to the scene is also a mesh, so test all the triangles to see if all are inside the room.

My idea was to do something similar to that. I thought about converting the mesh to a point cloud then perform a downsample and check the remaining points. 

Thinking a bit ahead, about checking the ground, my idea was to get the distance to the mesh pointing downwards and for each object have a pre-defined distance of the centroid to the lowest part of the object. With this, subtract the excess distance and set the pose of the object touching the ground. Is this overthinking the problem?

Sorry for my absence this week, I will work hard on this.

pmdjdias, 2023-03-06:
For the ground (I guess moving an object of the ground), I would simply fix the Y coordinate (height) so that the object is on the ground and move the object only in the X,Y coordinates (supposing Z is up).

andrefdre, 2023-03-06:
I followed your suggestions and used a convex hull otherwise it was too slow. This are the results:

https://user-images.githubusercontent.com/58526188/223153131-062d4325-facc-4ae3-96b4-fe96e261455f.mp4

I don't know if with more objects it wouldn't become too slow.

andrefdre, 2023-03-06:
I implemented random object placement, always touching the ground.

https://user-images.githubusercontent.com/58526188/223188528-d284868f-1f86-484e-8258-eb3f0903fea5.mp4

I don't understand why the chair doesn't always move, I confirmed it's generating the pose and going into the function that sets it. But it doesn't move, only after many tries, or I pause the simulation and give it time to move. Going to investigate more about this.

miguelriemoliveira, 2023-03-06:
> I implemented random object placement, always touching the ground.

This looks really nice. Congratulations.

Some ideas to make it better:

1. More models of humans, chairs and other plausible objects
2. Rotate the objects around the z axis (the human is always facing one direction)

> I don't understand why the chair doesn't always move, I confirmed it's generating the pose and going into the function that sets it. But it doesn't move, only after many tries, or I pause the simulation and give it time to move. Going to investigate more about this.

ok, I remember the gazebo services would sometimes fail with no explanation why...

andrefdre, 2023-03-06:
> 2\. Rotate the objects around the z axis (the human is always facing one direction)

It is rotating, the first 3 poses it's possible to see, the last two were just coincidentally the same.

> 1. More models of humans, chairs and other plausible objects

I will take one more day looking into the gazebo service to see if I can get to something, and afterward I will do this. Right now it's very easy to add more objects, the code is already implemented in a way to accept as many objects as previously defined.

miguelriemoliveira, 2023-03-06:
Great!

andrefdre, 2023-03-07:
I figured out why the objects weren't moving. In the SDF file, they had the property static set to true.

https://user-images.githubusercontent.com/58526188/223504306-fe635794-0145-4dbf-8e13-34bbaafd6ed4.mp4

I noticed sometimes objects spawn on top of the table or close to the camera, what should I do about this?

pmdjdias, 2023-03-07:
Regarding the camera, shoud be easy, you can define a threshold distance between object centroid and camera position and discard object that are below the threshold.
For the table it is harder, unless you manually define an exclusion area around the table. In my opinion we shoudl leave this possibility, the netwrok shoud learn also with this info, altough it is not common to walk over tables ;-)!

andrefdre, 2023-03-08:
> Regarding the camera, shoud be easy, you can define a threshold distance between object centroid and camera position and discard object that are below the threshold.

I followed your suggestion and also take into account the camera position. I placed a threshold distance of 0.5.

Is it possible to have a meeting this week to discuss the results and next steps?

miguelriemoliveira, 2023-03-08:
Sure. You advanced a lot. Friday 9h or 10h?

pmdjdias, 2023-03-08:
On friday I I have a meeting at 9:30 (and 11, and 12 ;-)). 10:00 would be better for me.

DanielCoelho112, 2023-03-08:
10h is good for me too.

andrefdre, 2023-03-08:
10h is good for me.

miguelriemoliveira, 2023-03-13:
> I couldn't find more models in the gazebo database, so I went looking for
> other alternatives.
>
> I managed to find this tutorial
> https://www.youtube.com/watch?v=7kEmT-NE75c. With the MakeHumans and
> blender programs I manage to create people and deploy them in gazebo. Now
> is just a matter of creating more.
>
> [image: image]
> <https://user-images.githubusercontent.com/58526188/224762246-ae58f208-b10d-4167-91e4-beceee830c2d.png>

Nice. A lot of people and furniture.

*  Miguel Riem de Oliveira        Professor Auxiliar         *
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

andrefdre, 2023-03-16:
I added more objects and more people. I have a total of 18 people and 6 objects.

Video 4x speedup:

https://user-images.githubusercontent.com/58526188/225718044-13a0292a-cf24-4f95-83a4-fba6385b5a5c.mp4

pmdjdias, 2023-03-17:
Pergunta: Verificas se existe colisões entre a camara e os objetos (como se faz com as paredes da sala). É que isso pode levar a ter imagens dentro dos objetos, até se se será sempre uma situação residual.

miguelriemoliveira, 2023-03-17:
I like the video. I don't like the camera model. It looks like a big exclamation point. How are you drawing it?

andrefdre, 2023-03-17:
> I like the video. I don't like the camera model. It looks like a big exclamation point. How are you drawing it?

During the meeting we talked about adding something to the camera, I thought about the exclamation since it was easy to do. I did it, basically adding a cube and a rectangle. The camera model is still bellow it. I will try to draw a camera warning sign to make it look better.

andrefdre, 2023-03-17:
I was more thinking maintaining the current kinect camera model and, on top, just adding a camera icon. Just to make it visible.

miguelriemoliveira, 2023-03-17:
Hi André,

Better to increase the model size.

But the frustration is even better,  I think.

DanielCoelho112, 2023-03-17:
Hi @miguelriemoliveira and @andrefdre,

we have a frustum but with marker arrays for the RViz. I don't think it will help here.

https://github.com/DanielCoelho112/synfeal/blob/d472d1c54deaa7f639a807c27d945f6a8792116d/utils_ros.py#L88

I would increase the camera's size and maybe change the color.

andrefdre, 2023-03-20:
> How about a bigger size?

This is just about aesthetic, right? If so, I honestly prefer having an icon than increasing the model size. But if I'm the only one, I will try a bigger model.

miguelriemoliveira, 2023-03-20:
Hi @andrefdre ,

its not just aesthetic. I mean, these are the images we are going to put in a paper, so they should be good at showing what we want to show. 

But I agree that we should not spend a lot more time on this.
Just one more suggestion: how about adding to the  model  of the camera a pyramid representing the view frustum?

@DanielCoelho112 can you post those images from RVIZ, its something like that ...

In any case you should proceed to the training of models.

andrefdre, 2023-03-20:
> how about adding to the model of the camera a pyramid representing the view frustum?

I think that's possible. Tomorrow I will post a picture of that implementation.

> In any case you should proceed to the training of models.

Are the objects enough? How many for test and how many for training? 

I also started working today on recreating what the state of the art currently use. I tried two approaches. 

First I tried using the same dataset, detect the objects and then place black boxes on them but had many false positives or not detecting some of them.

I then tried to read movement from a dataset with objects and replicate its camera movement. But the black boxes will be random.

Does the teacher have any suggestion?

miguelriemoliveira, 2023-03-20:
> Are the objects enough? How many for test and how many for training?

The video looks nice. I would try to train with these objects.
I would use 80% for training, and 20% for testing.

> Does the teacher have any suggestion?

Yes, the first is to create a new issue about this.

Can you do it and then we discuss there.

pmdjdias, 2023-03-21:
Nice!

DanielCoelho112, 2023-03-21:
@andrefdre 

wireframe would be nice. If possible, something like the ones in RViz (check the architecture from the paper).
[image]

miguelriemoliveira, 2023-03-21:
Yes, wireframe would be better. 

Also, the frustrum should start from the camera's pose, not the icon above.

Finally, the angles of the pyramid could be more realistic. @DanielCoelho112 can give you the correct angles, or you can search in the camera xacro.

andrefdre, 2023-03-21:
What does the teacher think @miguelriemoliveira, does it look better?

[image]

miguelriemoliveira, 2023-03-21:
Much, much better. I was worried for being such a  pain, but I am glad I
was.

It looks great! Congrats

pmdjdias, 2023-03-22:
Really nice, yes!

miguelriemoliveira, 2023-03-29:
Hi @andrefdre ,

this still  does not look good. I think the training is still not going well.

You should  run a training dataset without objects (using one of Daniel's datasets) and see if the training actually works.
I remember I suggested it before ... did you do it already?

pmdjdias, 2023-03-29:
I agree... you shoudl replicate Daniel's work with you dataset to vbe sure the whole pipeline is OK, and then after tray woth object occlusions!

andrefdre, 2023-03-29:
> You should run a training dataset without objects (using one of Daniel's datasets) and see if the training actually works.
> I remember I suggested it before ... did you do it already?

Yes, I already did that some weeks ago and it worked perfectly. I think these results are expected due to what we are asking the model to do. I spoke with @DanielCoelho112 and he suggested doing hyperparameter optimization.

miguelriemoliveira, 2023-03-29:
> Yes, I already did that some weeks ago and it worked perfectly. 

Can you please post here the graph of the loss training and test and give some indications on the size of the dataset and other relevant properties in both cases?

> I think these results are expected due to what we are asking the model to do.

what do you mean. Can you try to explain better?

>  I spoke with @DanielCoelho112 and he suggested doing hyperparameter optimization.

Not sure about that ... I mean, it is not easy to find a working combination of hyperparameters. @DanielCoelho112 found one which we know works without objects. Why don't we stick to it?
Inserting objects in the scene would not change the task significantly ... it may loose some accuracy, but not stop converging altogether.

DanielCoelho112, 2023-03-29:
Hi @andrefdre, @miguelriemoliveira, @pmdjdias,

> Can you please post here the graph of the loss training and test and give some indications on the size of the dataset and other relevant properties in both cases?

This was the curve I obtained in the sanctuary with 95k frames:
[image]

I believe @andrefdre already obtained similar results, but please share your curves.
At first sight, it looks that the model in the previous image is overtiffing. However, that is because of the loss function that is different from training and testing. In training, it is the sum of 2 negative quantities and in test, it is only one. 

This explains the difference between the train and test loss.

One thing @andrefdre  forgot to mention is that he changed the way we are plotting data. In my curve, as soon as the test loss starts increasing, I stop appending data to the plot. And that is why his test loss keeps increasing. I like this change, in fact, I am doing exactly the same thing in my projects.

In the plot shared by Andre, if we only observe the data until the blue vertical line, it seems close to what I obtained. 

> Inserting objects in the scene would not change the task significantly ... it may loose some accuracy, but not stop converging altogether.

The plot shared by Andre shows a model that is overfitting, which is different from converging. The model is converging to a minimum, we can see the training loss stabilizing.  About the overfitting, we are doing early termination (vertical blue line) and applying some regularization techniques. Since the task changed slightly, I think we should do a few more runs with different hyperparameters to see if we can minimize the overfitting. Possible changes could be: Increasing dropout, increasing weight decay, change Dynamic Loss parameters initiation.

@andrefdre, run the script to produce results on the test data so we can see the magnitude of the errors.  It is always good to evaluate the model with objective metrics besides the loss function. Especially when the loss function has learnable parameters.

miguelriemoliveira, 2023-03-29:
Ok, so the curve is not so bad.

Now I understand why @DanielCoelho112 suggests tuning the hyper parameters.

Another solution would be to get a dataset with less people in it. Say only one person per frame or something.
But I think you should try with this dataset for now.

andrefdre, 2023-03-29:
I don't have the results using @DanielCoelho112 datasets anymore, and unfortunately I already removed them from my folder in Deeplar to release space. But I already trained with the same path of the dataset of objects but without the objects and this is the results.

[image]

> Another solution would be to get a dataset with less people in it. Say only one person per frame or something.

But not always the objects spawns in front of the camera, so only one person would be very sparse.

> @andrefdre, run the script to produce results on the test data so we can see the magnitude of the errors. It is always good to evaluate the model with objective metrics besides the loss function. Especially when the loss function has learnable parameters.

As soon as I can, I will do this. Just to get registered what we talked, The validation dataset should be 24k images and use the same objects as the test, since the model never saw them.

miguelriemoliveira, 2023-03-30:
> But not always the objects spawns in front of the camera, so only one person would be very sparse.

I mean reduce the number of objects. Perhaps not to 1 per frame, but half of what you have now or something.
In any case I think @DanielCoelho112 's suggestion for now makes more sense.

andrefdre, 2023-03-30:
> @andrefdre, run the script to produce results on the test data so we can see the magnitude of the errors. It is always good to evaluate the model with objective metrics besides the loss function. Especially when the loss function has learnable parameters.

[image]

The mean error is:
 - Position: 0.24m
 - Rotation: 16.6º

The median error is:
- Position: 0.19m
- Rotation: 11.16º

andrefdre, 2023-03-30:
Results from the model using data augmentation done by state of the art:

[image]

The mean error is:

- Position: 0.20m
- Rotation: 15.4º

The median error is:

- Position: 0.17m
- Rotation: 10.13º

DanielCoelho112, 2023-03-30:
First of all, getting 0.19m and 11.6º is exactly what we were expecting. In the paper, using the same dataset size we obtained 0.12m and 7.91º. So we can say that the full pipeline is working!

About the results of the data augmentations, I'm a little surprised... I was really thinking the error would increase substantially. Can you give more information about the data augmentation used?  Something we can try is to add more objects to make sure they have some influence.

andrefdre, 2023-03-30:
> Can you give more information about the data augmentation used?

I followed your suggestion in the issue #95 and added `transforms.RandomErasing()` to the rgb_transofrm_train so the final result was:

```
        rgb_transform_train = transforms.Compose([
            transforms.Resize(args['crop_size'] + 1),
            transforms.RandomCrop(args['crop_size']),
            transforms.ToTensor(),
            transforms.Normalize(rgb_mean, rgb_std),
            transforms.RandomErasing()
        ])
```

miguelriemoliveira, 2023-03-30:
> About the results of the data augmentations, I'm a little surprised... I was really thinking the error would increase substantially. Can you give more information about the data augmentation used? Something we can try is to add more objects to make sure they have some influence.

I agree. We were expecting it to be worse.

> I followed your suggestion in the issue https://github.com/DanielCoelho112/synfeal/issues/95 and added transforms.RandomErasing() to the rgb_transofrm_train so the final result was:

Post a couple of images used in the training / evaluation of results using this method, just to get an idea.

andrefdre, 2023-03-30:
These are a set of images from a part of a batch during training. The colors are due to the normalization and the black squares are the random erasing.
 
[image]

miguelriemoliveira, 2023-03-30:
So they make sense right ...

miguelriemoliveira, 2023-03-30:
Lets speak tomorrow, at 9h or at 11h. Which hour is good for you?

andrefdre, 2023-03-30:
> Lets speak tomorrow, at 9h or at 11h. Which hour is good for you?

Any time is good but I would prefer at 9.

miguelriemoliveira, 2023-03-31:
Hi @andrefdre ,

Better at 11h for me.

andrefdre, 2023-03-31:
Okay sounds good for me.

DanielCoelho112, 2023-03-31:
I'll show up too.

andrefdre, 2023-04-01:
I finalized the script to generate video from a dataset, I added a parameter to choose the fps and the default is 60. Here are the results from the test dataset with 6 objects. I noticed that the camera sometimes behaves strangely when checking the collision with the mesh @DanielCoelho112 have you ever experienced this?

Video: https://www.youtube.com/watch?v=t53RIy-cIl0

miguelriemoliveira, 2023-04-01:
Hi @andrefdre ,

the functionality looks great. Some takeaways:

1. Most of the images look adequate for localization
2. One problem is that it only went once to the stairs
3. It seems the objects are in reasonable number
4. Objects are in reasonable places (except for one person on top of the table, but I think that's not important)

Just a question to be sure: The black boxes approach uses exactly the same images but without objects, right?

And a difficult question for @DanielCoelho112 looking at the images I realized that all the walls have unique textures. Could it be that this causes a boost in the localization performance which is actually the responsible for achieving 12 cm? I mean, if the walls were white, like I suspect they should be in the other datasets, it would be much harder to localize, even for smaller scenarios.

andrefdre, 2023-04-01:
> 2\. One problem is that it only went once to the stairs

Don't think that will present a problem, this is a small dataset, with a bigger with 95k it will have a higher chance of going there.

> 3\. It seems the objects are in reasonable number

I'm finishing the dataset with augmentation and I don't know if I agree, at least the black boxes are more common than the objects. Give me 1 hour to finish creating the video.

> Just a question to be sure: The black boxes approach uses exactly the same images but without objects, right?

They are supposed to be the same, but I'm reviewing and in the orientation there must be a bug somewhere, because it's tilted a bit down. It might be due to @DanielCoelho112 round the position and orientation, but I don't believe that myself. The position looks good.

andrefdre, 2023-04-01:
Video with data augmentation: https://www.youtube.com/watch?v=Yc1SLXjtLFo

miguelriemoliveira, 2023-04-02:
Hi @andrefdre ,

about the video of the black boxes approach: it appears to be working fine. 

about the results: it is good that the approach trained with no objects is worse on an objects  test dataset when compared to our approach of training with objects. That should be the case.

I don't think we are out of the woods just yet, since the black boxes approach is our data augmentation competitor and appears to work much better (Position 0.017m right?).

andrefdre, 2023-04-02:
> about the video of the black boxes approach: it appears to be working fine.

It works fine, but clearly there is a difference in the percentage of black boxes vs objects present in each dataset, maybe we should try having more objects, right?

> I don't think we are out of the woods just yet, since the black boxes approach is our data augmentation competitor and appears to work much better (Position 0.017m right?).

Yes, but now that I trained again with augmentation and just changed the test dataset to be with objects, it had worse performance. It could be due to one saved at 20 epochs (the better results one) and the other saved at 10 due to higher loss in the test dataset. Or maybe it could be what Daniel said, that these values are small and that training and extracting results one time are not significant enough.

[9 more replies]

closed 2023-12-12 by DanielCoelho112