DanielCoelho112/synfeal #96: Real time factor in Gazebo decreases to low values (0.05) during dataset acquisition

andrefdre, 2023-03-24:
I started recording dataset to train, and I left the computer all night to find in the next day that it only had taken 24k images. I noticed that the Real time Factor had decreased. After some tests and a bit of research, I couldn't find a culprit. My best guess is due to the object's meshes, since when I remove them, the simulation goes to normal. But when the code adds the objects again, it gets again slow.
I also noticed that in the beginning the simulation is fast, but it starts to decrease over time to where around 5k images it's half speed. I tried disabling physics simulation since we don't need it and had a slight improvement, but nothing significant. With a Real Time Factor of 0.05 the simulations becomes unusable for large datasets. 
@DanielCoelho112 suggested to record small datasets and then join them. Does @miguelriemoliveira have any suggestion?

miguelriemoliveira, 2023-03-24:
Sounds like a memory leak. I will call

andrefdre, 2023-03-24:
In case it's a memory leak from deleting the models and spawning them, an idea could be to spawn all and when not in use move to a position else where.

miguelriemoliveira, 2023-03-24:
Right, good idea.

And also try to find a way to reset the model in gazeebo...

andrefdre, 2023-03-26:
Yesterday I tried to find if I could find a memory leak. I recorded the memory usage for the gazebo (client and server) and also the python script. I don't think I found anything, but I don't have much experience in this. Here are the results:

[image]

While I was searching about this problem I found this issue https://github.com/gazebosim/gz-gui/issues/208, and they found a memory leak, but it's for ignition gazebo. Apparently deleteModel service doesn't delete the collision and visual mesh, so even though it's not being rendered it's still taking into account when simulating physics.

I don't think it's worth it losing more time trying to find where the issue comes from, since when I implemented, the idea of just moving the objects to a resting position seems to solve the problem. For this, I had to remove all the collision meshes in the SDF, which aren't needed since we verify everything in the python script. Not using a collision mesh also leads to using less memory, which might be good. 

I also changed the real time update parameter to 2000, so the simulation goes 2x the speed, making it faster to generate the dataset.

Another thing I did was to calculate the object's position with multiprocessing, so it generates new positions during image acquisition

andrefdre, 2023-03-26:
Yesterday I tried to find if I could find a memory leak. I recorded the memory usage for the gazebo (client and server) and also the python script. I don't think I found anything, but I don't have much experience in this. Here are the results:

[image]

While I was searching about this problem I found this issue https://github.com/gazebosim/gz-gui/issues/208, and they found a memory leak, but it's for ignition gazebo. Apparently deleteModel service doesn't delete the collision and visual mesh, so even though it's not being rendered it's still taking into account when simulating physics.

I don't think it's worth it losing more time trying to find where the issue comes from, since when I implemented, the idea of just moving the objects to a resting position seems to solve the problem. For this, I had to remove all the collision meshes in the SDF, which aren't needed since we verify everything in the python script. Not using a collision mesh also leads to using less memory, which might be good. 

I also changed the real time update parameter to 2000, so the simulation goes 2x the speed, making it faster to generate the dataset.

Another thing I did was to calculate the object's position with multiprocessing, so it generates new positions during image acquisition is it possible this leads to memory leak? It doesn't seem like it.

miguelriemoliveira, 2023-03-26:
> While I was searching about this problem I found this issue https://github.com/gazebosim/gz-gui/issues/208, and they found a memory leak, but it's for ignition gazebo. Apparently deleteModel service [...]

I read through the issue. Apparently its not solved yet.

> I don't think it's worth it losing more time trying to find where the issue comes from, since when I implemented, the idea of just moving the objects to a resting position seems to solve the [...]

Ok, sounds good.

> I also changed the real time update parameter to 2000, so the simulation goes 2x the speed, making it faster to generate the dataset.

Nice.

> Another thing I did was to calculate the object's position with multiprocessing, so it generates new positions during image acquisition is it possible this leads to memory leak? It doesn't seem like it.

Not sure about it. In any case you can now produce a bigger dataset faster, and you also have the option to merge two datasets right? I would say move forward to producing the datasets and to training.

andrefdre, 2023-03-26:
> Not sure about it. In any case you can now produce a bigger dataset faster, and you also have the option to merge two datasets right? I would say move forward to producing the datasets and to training.

I haven't yet tried @DanielCoelho112 code to merge datasets, but from what he said, it's possible to merge. Now, I will move on to generate dataset and get results.

closed 2023-03-26 by andrefdre

pmdjdias, 2023-03-27:
I've been through the issues and it appears the thing are going (although wth some bumps along the road ;-)).
Looking forward to sse some results of the training with datasets.