DanielCoelho112/synfeal #98: Time limitations to train all the models

andrefdre, 2023-05-04:
Due to time limitations, it's not possible to train 5 models for each case and network. Currently, I only have available 1 GPU in deeplar. What should I do, @miguelriemoliveira?

DanielCoelho112, 2023-05-04:
As I previously suggested, I think that one run for each case and network is sufficient.  The tables show that there are no significant differences between runs.

miguelriemoliveira, 2023-05-04:
Right. Let's do 1 run per approach. Later if we have more gpus we can go back to this ...

miguelriemoliveira, 2023-05-04:
@DanielCoelho112 , can't we give some priority to @andrefdre  since he's close to finishing the thesis?

DanielCoelho112, 2023-05-04:
> @DanielCoelho112 , can't we give some priority to @andrefdre since he's close to finishing the thesis?

Yes, we can. I'll liberate one GPU by the end of the day.
Still, I think one run is enough. All papers in localization (and in Deep Learning in general) are only trained once. Running 5 runs, would take around 50 experiments, which would take too much time.

miguelriemoliveira, 2023-05-04:
Right. One is enough. Just i  case he needs it for some other test...

andrefdre, 2023-05-05:
Thanks, I will then train only one model with 2 GPU I will end the trains faster.

closed 2023-05-05 by andrefdre